timofurrer/w1thermsensor #10: Feature request: add support to allow changing resolution

CptanPanic, 2015-08-11:
Maybe this module is too high level since it uses w1_therm but since w1_therm doesn't allow you to change resolution, it would be great if this library allowed you to query and change resolution.

timofurrer, 2015-10-07:
What do you mean with "change the resolution" ?

CptanPanic, 2015-10-07:
The DS18B20 allows for adjusting the returned temp resolution from 9 - 12 bits.  Higher resolution vs. slower results.

timofurrer, 2015-10-07:
Okay, I think I'll not be able to change the resolution on kernel model side (anyway I'll have a look at it). I can definitely make the resolution changeable for the module module, but of course only in the range which is provided by the kernel module. Do you have some ideas how you want to specify the resolution so it fits your needs best?

srwareham, 2016-02-15:
This issue is still open and I agree it would be a great feature so I thought I might step in with a response.

Related to setting the resolution, it would probably easier to first think of the problem of outputting the current resolution.  As the resolution for the DS18B20 is inextricably linked to its refresh rate, I think this information would best be served in a tuple of (resolution_in_degrees_celsius, minimum_refresh_rate_in_seconds).

Going back to the original question of setting values, I believe each device could have a list of such tuples that the device can accommodate. The user could then obtain this list, and decide which configuration is desired (likely through sorting by resolution, or by refresh rate).

As the implementation would be related, my questions would be whether or not it is possible for multiple devices of the same type to use different kernel modules at the same time. Similarly, whether or not different devices could be used with different kernel modules at the same time. I would defer these questions to you to determine the abstraction level at which getting and setting should occur.

Do you think these features might make it in? Thanks for the great work!

Sources:
http://www.homautomation.org/2015/11/17/ds18b20-how-to-change-resolution-9101112-bits/)
https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf

timofurrer, 2016-03-04:
If this would work I'd be more than happy to integrate them. I have to dig into this in order to give a more detailed answer. If anyone could make a prototype or a step forward I'd be happy to look at it and maybe integrate it!

lloydjatkinson, 2016-04-17:
Hi,

If you read the documentation/source (https://www.kernel.org/doc/Documentation/w1/slaves/w1_therm )for w1-therm the following can be found:

> The driver also doesn't support reduced precision (which would also reduce the conversion time).

This matches up with what @CptanPanic said.

It's worth mentioning that the resolution bits are stored in EEPROM on board the devices. 

>  The configuration register allows the user to set the resolution of the temperature-to-digital  onversion to 9, 10, 11, or 12 bits. The TH, TL, and configuration registers are nonvolatile (EEPROM), so they will retain data when the device is powered down.

https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf

So a potential work around w1-therm's lack of resolution changing support, you could manually change the resolution of each sensor using some other method. **Then** connecting the sensor back to whatever you are using to read the temperature. A better solution would be for the w1-therm to have resolution support added.

flaggz, 2017-02-17:
already in next rpi kernel: https://github.com/raspberrypi/linux/pull/1412
could be easily added in this library after the kernel goes mainstream

timofurrer, 2017-02-17:
Thanks @flaggz ! That's great.

I've implemented a first prototype at #29 .

Does any one want to double check if the current implementation works?

I still have to do thorough testing and implement the CLI command.

ComputerSmiths, 2017-05-15:
I must be dense, I tried:

from w1thermsensor import W1ThermSensor

for sensor in W1ThermSensor.get_available_sensors():
        print("Sensor %s has temperature %.2f" % (sensor.id, sensor.get_temperature()))
        print("Setting to 12-bit precision...")
        sensor.set_precision(12,True)

but I get:

Sensor 0000080143e7 has temperature -24.62
Setting to 12-bit precision...
Traceback (most recent call last):
  File "test1w.py", line 6, in <module>
    sensor.set_precision(12,True)
AttributeError: 'W1ThermSensor' object has no attribute 'set_precision'

What's the proper incantation?

Thanks!

timofurrer, 2017-05-15:
@ComputerSmiths Which branch did you try? #29 is not released yet. I'll wait with the merge until Kernel 4.7 is in raspbian by default.

ComputerSmiths, 2017-05-15:
Ah, that must be my problem.  That is the proper incantation, though?

timofurrer, 2017-05-15:
`incantation` ?

ComputerSmiths, 2017-05-15:
I can't find any documentation, is the call to:

sensor.set_precision(12,True)

the appropriate one? Great API, BTW, many thanks for making it available!

timofurrer, 2017-05-15:
Yes, that's the proper call if you want to persist the precision in the EEPROM of your sensor. 
If you don't want that you should pass `False` for the second argument.

ComputerSmiths, 2017-05-15:
Thanks, I'll wait for Raspian kernel 4.7

bsimmo, 2017-05-16:
Raspian is at kernel 4.9 by default!

bsimmo, 2017-05-16:
Although you cannot guarantee everyone will be at 4.9, it's only been there for a fortnight.

timofurrer, 2017-05-16:
Thanks for the update! I'll have a look at it.

ComputerSmiths, 2017-05-16:
Hmm:
root@chillypi:~# uname -a
Linux chillypi.geekho.com 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux
root@chillypi:~# 

That means I'm running 4.9.24, yes?  Do I need to do something special to update w1thermsensor other than:

root@chillypi:~# apt-get install python-w1thermsensor
[...]
python-w1thermsensor is already the newest version.

root@chillypi:~# apt list python-w1thermsensor
Listing... Done
python-w1thermsensor/stable,now 0.3.3 all [installed]

I did:

git clone https://github.com/timofurrer/w1thermsensor.git
cd w1thermsensor/
python setup.py build
python setup.py install

but that didn't help...

timofurrer, 2017-05-16:
Yeah, in order to install it with `pip` or `apt` I have to release first - which I will do until tomorrow. 

If you want to use the WIP branch you can use it by cloning the repo, checking the branch to `feature/change_resolution` and install it (like you did with the master branch already).

But as I've said - it's still work in progress - so I don't guarantee that it works until it's released.

timofurrer, 2017-05-16:
I've just triggered the release for **w1thermsensor v0.4.0**.

If you run into any problems please open a new issue.

Thanks to all of you! :tada:

commit ee35e5e2 (2017-03-23): Implement command to change sensor precision. Closes #10
commit 3c436c0b (2017-05-16): Implement command to change sensor precision. Closes #10
commit 934a8c1a (2017-05-16): Merge pull request #29 from timofurrer/feature/change_resolution

Implement command to change sensor precision. Closes #10
closed 2017-05-16 by timofurrer