timofurrer/w1thermsensor #17: Onion.io compatibility

WattsC-90, 2016-03-04:
Nice library, what would i need to do to make this compatible with an onion.io board? Or would it be compatible out the box? Unsure how it handles the lowest level interactions with gpio

timofurrer, 2016-03-04:
The `w1thermsensor` module is using the `w1therm` kernel module. Thus, the low level interaction with the GPIO pins is handled by this kernel module and not by this python module. I'm pretty confident that if the kernel module works, this module will work too.
Unfortunately, I don't have any experiences with onion.io

Keep me posted if you try it!

timofurrer, 2017-02-13:
I'm going to close this issue until there is a concrete problem with this issue.

If you have any updates on the *onion.io* compatibility I'm happy to here all about it - just use this issue to respond.

closed 2017-02-13 by timofurrer