timofurrer/w1thermsensor #18: RPi, which input to use?

WattsC-90, 2016-03-26:
Hi, 
me again!
Just trying to familiarise myself with this library, but getting no sensor found error, which input should the sensor be connected to on an rpi zero? Is there any way to change that pin? 
Thanks

timofurrer, 2016-03-26:
Unfortunately, I don't have any experiences with the pi zero. However, this python package is not responsible for detected the sensors from the connected pins. This is what the `w1-gpio` kernel module is for. It basically detects the sensors and creates some files which are then read by this python package. 

What's the output of `ls -lR /sys/bus/w1/devices`?

WattsC-90, 2016-03-29:
Hi Timo,
thanks for getting back to me. I was trying to establish which pin you used to connect into etc. on the RPi. There is conflicting information on the net. 
The output of the ls command varies, showing the 1w-bus-master and 00-2000 etc, 00-4 etc.
I did realise that, after asking originally, I hadnt enabled the w1-gpio module in the pi, just trying to establish the hardware connections to figure out whats going on.

timofurrer, 2017-02-13:
>  did realise that, after asking originally, I hadnt enabled the w1-gpio module in the pi, just trying to establish the hardware connections to figure out whats going on.

Any news on this one?

petonic, 2017-03-15:
On the RPI0 or any, really, it uses GPIO 4 (Pin 7).  Same as RPI3.

Thanks, Timo, for writing this and making it available to the community.  It's a really solid, simple, package 👍

timofurrer, 2017-03-15:
> On the RPI0 or any, really, it uses GPIO 4 (Pin 7). Same as RPI3.

Thanks for the input!

> Thanks, Timo, for writing this and making it available to the community. It's a really solid, simple, package :+1:

And thanks for the feedback! I appreciate it!

@WattsC-90 I assume you solved the problem by enabling the kernel module?

bsimmo, 2017-03-15:
That's not quite right.
GPIO is the default pin used, but you can change it in /boot/config.sys file with a dtoverlay=w1-gpio settings
You can use raspi-config to enable it at it's default setting and it places it there for you
You should not need to do anything after that and everything runs as it should, no modprobes are needed as the overlay does it for you.

marcosadsj, 2017-04-25:
@bsimmo 
I tryied all options, as change /boot/config.sys and others tutorials, but only worked it change in the raspi-config, through your comment! THANKS!

timofurrer, 2017-05-16:
This question seems to be answered, thus I'm going to close this.

I might create a little FAQ and link to this issue since it seems to be important. 

Thanks :tada:

closed 2017-05-16 by timofurrer