timofurrer/w1thermsensor #19: Error on CLI

allusionist, 2016-04-18:
Using Python 2.7.9 on latest Raspbian Jessie on a Pi 3. Installed via `pip install w1thermsensor`.
Attempting to use the CLI results in a "no such file or directory" error regarding 'test/mockedsensors'.

Kernel modules `w1-gpio` and `w1-therm` are loaded. A single MAX31850 is connected to GPIO 4. (IC GND - Pi GND; IC VIN - Pi 3.3V; IC 3.3V - 4.7K Resistor - IC DATA and Pi GPIO 4).

Running `w1thermsensor` in the command line with no arguments functions normally.

Running `w1thermsensor ls` or `w1thermsensor all` results in this error message (same for both):
```
Traceback (most recent call last):
  File "/usr/local/bin/w1thermsensor", line 9, in <module>
    load_entry_point('w1thermsensor==0.3.0', 'console_scripts', 'w1thermsensor')()
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/cli.py", line 87, in all
    sensors = W1ThermSensor.get_available_sensors(types)
  File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py", line 94, in get_available_sensors
    return [cls(cls.RESOLVE_TYPE_STR[s[:2]], s[3:]) for s in listdir(cls.BASE_DIRECTORY) if is_sensor(s)]
OSError: [Errno 2] No such file or directory: 'test/mockedsensors'

```

Running `w1thermsensor get (any valid argument)` results in this error message instead:
```
Traceback (most recent call last):
  File "/usr/local/bin/w1thermsensor", line 9, in <module>
    load_entry_point('w1thermsensor==0.3.0', 'console_scripts', 'w1thermsensor')()
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/cli.py", line 135, in get
    sensor = W1ThermSensor(type_, hwid)
  File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py", line 112, in __init__
    self._load_kernel_modules()
  File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py", line 155, in _load_kernel_modules
    raise KernelModuleLoadError()
w1thermsensor.core.KernelModuleLoadError: Cannot load w1 therm kernel modules
```

Thanks for any help you can provide!

timofurrer, 2016-04-18:
Could you please test with the master if the first error still happens?

I'll need to investigate asap.

jasonehines, 2016-04-24:
I was getting this too. It looks like the Rasberian file doesn't have your last commit.
 
I renamed all the `.pyc` files in `/usr/local/lib/python2.7/dist-packages/w1thermsensor` and then commented out line 13 of `cli.py` where it says `W1ThermSensor.BASE_DIRECTORY = "test/mockedsensors"`

timofurrer, 2016-04-25:
I just released `0.3.1`. It should be fixed there.

allusionist, 2016-04-25:
It appears to be working on my end now as well, though at the moment I don't have a TC on hand to read. Thank you!

closed 2016-04-25 by allusionist