timofurrer/w1thermsensor #20: Rasbian apt-get install not working(?)

archonic, 2016-04-24:
I'd rather not install pip if I don't have to. I'm trying to keep my SD image slim by using Minibian. I've run `sudo apt-get install python-w1thermsensor` and it seems to have installed just fine, but there's no `w1thermsensor` command available. `python-w1thermsensor` also doesn't work. I also tried reloading bash and rebooting.

timofurrer, 2016-04-25:
That's true. The new version of `python-w1thermsensor` is not available in the raspbian apt repositories yet. I'm working on it to get the new version in there.

archonic, 2016-04-25:
Ah that explains it. I ended up installing python3 and python3-pip for a recent pymongo anyways. Ultimately I'm going for a script which dumps sensor ids, temperature and timestamps to Mongo. If answering that interests you, I've got a stackexchange question here: http://raspberrypi.stackexchange.com/questions/46373/performant-way-to-dump-ds18b20-data-to-mongo

closed 2016-04-25 by archonic

timofurrer, 2016-04-25:
Oh, that sounds nice :) I'll have a look at it. I would also be interested in the *example gist*.

:beers:

archonic, 2016-04-25:
Here's the example gist:
https://gist.github.com/archonic/8d2d936c8f25f3cb84004f01d58c135c

I think Meteor might be super over-built for my purposes (brewing beer). I may just use Flask, a websocket package and tail a log file.

risoul, 2016-10-05:
Installed today with apt-get and same issue

timofurrer, 2016-10-05:
Only the `python3-w1thermsensor` package comes with the CLI tool - I have to state this in README.