timofurrer/w1thermsensor #25: Interference with RPi.GPIO/RPi.PWM

csmotion, 2016-10-09:
Timo,

Thanks for the excellent library!  

I'm trying to use this with the RPi.PWM library, but it appears that I run into problems if I run RPi.GPIO and w1thermsensor in separate processes.  However, if I roll these into a common module, there appears to be no issue.  Have you observed this behavior, and if so, thoughts on its cause?

Thanks!
Jim

timofurrer, 2016-11-18:
No, I haven't observed this behavior. 

>  I run into problems

Problems like what? Tracebacks? logic errors? wrong temperatures?

Thanks for your contribution :beers: And sorry for being late with answering!

csmotion, 2016-12-08:
Timo,

This appeared to be an issue with multiprocessing and RPi.GPIO, not your library!  D'oh!  Owe you a beer if you're in CO!

Jim

timofurrer, 2016-12-08:
Okay, no worries! I'm going to close this issue then!

I'll remember the beer though!

closed 2016-12-08 by timofurrer