timofurrer/w1thermsensor #28: w1thermsensor

petert100, 2016-12-05:
Hello, 
Great library.    I have used it successfully for groups of sensors. 
Something I have noticed and been unable to resolve: 
---------------------------
from w1thermsensor import W1ThermSensor
os.system('modprobe w1-gpio')
os.system('modprobe w1-therm')
while True:
    for sensor in W1ThermSensor.get_available_sensors():
        print("Sensor %s has temperature %.2f" % (sensor.id, sensor.get_temperature()))
   time.sleep(10)
-------------------------------------------------
the above works fine and if I run the program with 3 sensors I get a line for each.  
If I then stop the program and  remove  a sensor and run it again it returns this message :
"w1thermsensor.core.SensorNotReadyError: Sensor is not yet ready to read temperature"

What do I need to do to clear the old list of sensors and start over. ?

thanks

timofurrer, 2016-12-05:
This error occurs if the `w1therm` kernel module still knows the sensor which was once attached and with that there is still a file present in `/sys/bus/w1/devices` for that sensor but this file is not ready to be read - which is actually the expected behavior since the sensor is detached. So good so far.

In your case it might be enough to just `except` that exception and _ignore_ it.

If you do not want to _ignore_ this exception you could try to unload the kernel modules with `modprobe -r w1-therm w1-gpio`. And run your program again.

petert100, 2016-12-05:
That is kind of what I suspected.  
What should I use to ignore the error and where should I put it. ? 
thanks.

timofurrer, 2016-12-05:
Just a normal `try/except` block would do it. Something like:

```python
try:
    print("Sensor %s has temperature %.2f" % (sensor.id, sensor.get_temperature()))
except w1thermsensor.core.SensorNotReadyError:
   print("Sensor %s is not ready to read")
```

timofurrer, 2016-12-05:
I close this issue since it's not really one. But feel free to comment further - I'll help.

closed 2016-12-05 by timofurrer

petert100, 2016-12-06:
Thanks for the advice.

 here is what I got with the try/except as above

Sensor 000005302180 has temperature 21.00
Traceback (most recent call last):
  File "test_many.py", line 25, in <module>
    except w1thermsensor.core.SensorNotReadyError:
NameError: name 'w1thermsensor' is not defined

What am I missing?
peter

timofurrer, 2016-12-06:
You didn't import the `w1thermsensor` module - You did `from w1thermsensor import W1ThermSensor` at the top which only imports the `W1ThermSensor` class from the `w1thermsensor` package.

Replace it with `from w1thermsensor import W1ThermSensor, SensorNotReadyError` and change the `except` line to: `except SensorNotReadyError:`.

These are some really basic python things, so I suggest you to go through some tutorials to learn those. You'll see - it won't be hard at all. :beers:

petert100, 2016-12-06:
Timo,
Thank you very much.  Those things make sense.
M usual Python goto guy  ( my son)  is en route via the Alaska Highway  to
Anchorage and is out of touch.  so I really appreciate your taking time to
help me.

thanks again.    Peter

petert100, 2017-02-21:
Hello Timo 
I have another question I hope you can help me with. 
Is there a way(method?) to get the temperature from one known sensor by using the sensor-id?
Thanks
Peter

timofurrer, 2017-02-22:
Hi,

Checkout the README here: https://github.com/timofurrer/w1thermsensor#basic-usage-with-one-sensor-explicit

natanbenchimol, 2019-06-18:
Hey,

I've used this package before and really loved how easy it was to use. 
I am having issues downloading the package this time. When I try to install it on Rasbian using 
_sudo apt-get install python3-w1thermsensor_
I get the error: "unable to locate package python3-w1thermsensor"
I am using a raspberry pi with rasbian.

petert100, 2019-06-18:
As I recall, for me it turned out to have to do with the location of
Python. I was , still am, a newbie  and had several versions of it in
distinct locations.

After resolving that the package has worked very well for years!.....
Peter

timofurrer, 2019-06-18:
@natanbenchimol you've already opened an issue for your problem #55 , let's keep this one clean :) 

Thanks @petert100 and @natanbenchimol 🎉