timofurrer/w1thermsensor #32: Error: no such option --precision

Hornet-C, 2017-06-01:
Hi, thanks for the great lib, but I get the Error:

> w1thermsensor get 1 --precision 10
> Error: no such option: --precision

Sensor details:

> w1thermsensor ls
> Found 1 sensors:
>   1. HWID: 000008fdb09d Type: DS18B20

also 

> w1thermsensor get --hwid 000008fdb09d --type DS18B20 --precision 11

as well as the example code from python doesn't work for setting the precision.

System info:

> raspberrypi 3
> uname -a
> Linux raspberrypi 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux

timofurrer, 2017-06-01:
What version of the package are you using?
If you've installed the package via `apt` - The required changes are not deployed in the raspbian sources yet.

Hornet-C, 2017-06-01:
well okay, I've taken the raspbian sources via apt install. I'll try the sources directly from the git repo. Thank you very much.

Hornet-C, 2017-06-01:
Okay, the version from git seems to know the command, but I still get errors:

> w1thermsensor get 1 --precision 11
> Traceback (most recent call last):
>   File "/usr/local/bin/w1thermsensor", line 9, in <module>
>     load_entry_point('w1thermsensor==0.4.1', 'console_scripts', 'w1thermsensor')()
>   File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 722, in __call__
>     return self.main(*args, **kwargs)
>   File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 697, in main
>     rv = self.invoke(ctx)
>   File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 1066, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 895, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 535, in invoke
>     return callback(*args, **kwargs)
>   File "/usr/local/lib/python3.4/dist-packages/w1thermsensor/cli.py", line 146, in get
>     sensor.set_precision(precision, persist=False)
>   File "/usr/local/lib/python3.4/dist-packages/w1thermsensor/core.py", line 292, in set_precision
>     precision, self.sensorpath))
>   File "/usr/lib/python3.4/subprocess.py", line 537, in call
>     with Popen(*popenargs, **kwargs) as p:
>   File "/usr/lib/python3.4/subprocess.py", line 859, in __init__
>     restore_signals, start_new_session)
>   File "/usr/lib/python3.4/subprocess.py", line 1457, in _execute_child
>     raise child_exception_type(errno_num, err_msg)
> FileNotFoundError: [Errno 2] No such file or directory: 'echo 11 > /sys/bus/w1/devices/28-000008fdb09d/w1_slave'

The file exists:

> cat /sys/bus/w1/devices/28-000008fdb09d/w1_slave
> bb 01 4b 46 7f ff 05 10 c6 : crc=c6 YES
> bb 01 4b 46 7f ff 05 10 c6 t=27687

timofurrer, 2017-06-01:
What are the permissions on that file and with which user are you running the `w1thermsensor` process?

Hornet-C, 2017-06-01:
> ls -lh /sys/bus/w1/devices/28-000008fdb09d/w1_slave
> -rw-r--r-- 1 root root 4,0K Jun  1 17:58 /sys/bus/w1/devices/28-000008fdb09d/w1_slave

as user root.

I've installed the version after `git-clone ...` with `pip3 install .`

timofurrer, 2017-06-01:
I've just released a new version **v0.4.2**. You can also use pip directly instead of cloning the source first:

```
sudo pip3 install w1thermsensor -U
```

Can you try again?

Hornet-C, 2017-06-01:
Thanks, works great now !

closed 2017-06-01 by Hornet-C

timofurrer, 2017-06-02:
Awesome!

Thanks for the report! :beers: