timofurrer/w1thermsensor #40: w1thermsensor get -p 12 needs to be root

klint-k, 2017-09-01:
/bin/sh: 1: cannot create /sys/bus/w1/devices/28-0316b56fb3ff/w1_slave: Permission denied
Traceback (most recent call last):
  File "/usr/bin/w1thermsensor", line 9, in <module>
    load_entry_point('w1thermsensor==1.0.3', 'console_scripts', 'w1thermsensor')()
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.4/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/w1thermsensor/cli.py", line 145, in get
    sensor.set_precision(precision, persist=False)
  File "/usr/lib/python3/dist-packages/w1thermsensor/core.py", line 280, in set_precision
    raise W1ThermSensorError("Failed to change resolution to {0} bit".format(precision))
w1thermsensor.errors.W1ThermSensorError: Failed to change resolution to 12 bit

timofurrer, 2017-09-01:
Yes, it does need root permissions.
Running the command with `sudo` should work perfectly on a standard raspbian.

Did you try?

klint-k, 2017-09-01:
Hello,

 

Yes I tried.

This python script is past what I know, is there an error catch that will remind me to use sudo?

 

 

 

 

Klint

timofurrer, 2017-10-14:
The doc string of the `W1ThermSensor.set_precision` method already noted, that you might have to be root to change the precision.

In addition I've added a hint to the exception message and released **v1.0.5**.

Thanks :beers:

closed 2017-10-14 by timofurrer

klint-k, 2017-10-17:
Thanks for your help :)
I hope that I helped.
Klint

commit 714c5b1c (2017-10-14): Add need-root hint to set precision method. Closes #40