timofurrer/w1thermsensor #41: Threading of sensor reads

bsimmo, 2017-10-06:
As far as I can tell, from use, you access each sensor sequencially.
Would it be possible to implement threading to access them concurrently, this would speed up reading taking for more than one sensor.

I assume people would want to be able to disable the use of threads for whatever reason.

I've had a search and found this that may or may not give an idea.
https://gist.github.com/RobThree/c6b9a5dc16357eeb0e011d3ed9a281b0

timofurrer, 2017-10-06:
> As far as I can tell, from use, you access each sensor sequencially.

That's right. The `w1thermsensor` module does not interact with anything related to threading.

> Would it be possible to implement threading to access them concurrently

You can certainly do that. There should be nothing in the code which would restrict it for single-threaded use.

I'm not sure what you want exactly. Can you provide an example of something you want to achieve but doesn't work for you? That would help me to understand what you actually want to achieve  :)

bsimmo, 2017-10-06:
At the command line when I use
w1thermsensor all
it is done one after the other.

or in Python with the example
from w1thermsensor import W1ThermSensor
for sensor in W1ThermSensor.get_available_sensors():
____ print("Sensor %s has temperature %.2f" % (sensor.id,
sensor.get_temperature()))

that loop waits on the the sensor to return it's value.  I guess
.get_temperature() needs to be spwaned in the background module as a thread?
Though I guess if I figure out threading, I may be able to do it myself in
the loop now that I look at it.  It's been a while, since I already use
threading to write files and collect data.

It was just an idea to have it 'built-in'.

timofurrer, 2017-10-06:
Okay, I think I've got what you try to achieve. 
However, I want `w1thermsensor` to be as simple as possible and therefore want to omit every kind of threading by default. 

If you want multiple sensors to be read in parallel you can write a little wrapper around `w1thermsensor`.

I'm going to close this issue since multi-threading is definitely not in the scope of this project.

closed 2017-10-06 by timofurrer

bsimmo, 2017-10-08:
No problem, I guess i'll clone and try work it out myself :-)