timofurrer/w1thermsensor #42: Cannot load w1 therm kernel modules on Raspberry Pi

sandhan26, 2017-11-11:
On Raspbian Stretch, i am facing this issue.

Python 2.7.13 (default, Jan 19 2017, 14:48:08) 
[GCC 6.3.0 20170124] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from w1thermsensor import W1ThermSensor
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/__init__.py", line 7, in <module>
    from .core import W1ThermSensor  # noqa
  File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py", line 316, in <module>
    load_kernel_modules()
  File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py", line 310, in load_kernel_modules
    raise KernelModuleLoadError()
w1thermsensor.errors.KernelModuleLoadError: Cannot load w1 therm kernel modules

cyrisX2, 2017-11-13:
You need to update your boot file.
`sudo nano /boot/config.txt`

Then at the end add 
`dtoverlay=w1-gpio`

Finally, reboot 
`sudo reboot`

timofurrer, 2017-11-13:
@cyrisX2 Thanks for the info!

@sandhan26 did this work for you?

Bernard26, 2017-11-13:
Hi,
How to use different  GPIO pin for  DS18B20 data ?
Thank you

cyrisX2, 2017-11-14:
When you add the line to config, you can add the pin as well - default is 4. Just replace 'x' with whatever pin you want.

`dtoverlay=w1-gpio,gpiopin=x`

Bernard26, 2017-11-14:
Thank you very much **cyrisX2**

timofurrer, 2017-11-15:
I'm going to close this for now as it seems all questions are answered. @sandhan26 if it doesn't work for you, please let us know and I'll reopen this issue.

Cheers :beers:

sandhan26, 2017-11-18:
@timofurrer Sorry, for my delayed response. Let me elaborate first.

On my fresh installation of Raspbian Stretch Desktop, the BME280 sensor was working perfectly, with W1ThermSensor library. 

But, after few days of lying dormant, i tried to use the sensor again with W1ThermSensor for some other project again. And this time it gives me the above error.

I had already tried what others have suggested, but the issue still persists.

this is the **config.txt** file with the modification you all suggested.

[image]

Here are the outputs of `i2cdetect -y 1` and `w1thermsensor ls` 

[image]

As you can see, the address of the BME280 is shown on the i2c port of Raspberry Pi as 76, which rules out the possibility of incorrect wiring.

timofurrer, 2017-11-18:
Okay, that's unfortunate. 

Can you provide:
* the version of w1thermsensor you are using?
* did you have root permissions when you've got the traceback posted in the first post?
* the output of `ls /sys/bus/w1/devices`?

bsimmo, 2017-11-18:
Have you wired the DS correctly and the 4k7 resistor.
I2C has nothing to do with it so ignore the BME.
Also gpio 4 is the default pin.
Normally I would use raspi-config to enable it unless anything specific is
needed.

sandhan26, 2017-11-19:
@bsimmo yeah, the BME280 is wired correctly. And i did not use 4k7 resistor, as it was working without any resistors previously.

I did not understand your point "I2C has nothing to do with it so ignore the BME"

sandhan26, 2017-11-19:
@timofurrer 

1. I am using python3-w1thermsenso v1.0.3
2. I was logged into the default account 'pi'
3. I get the following list of folders
`00-478000000000  00-c78000000000  w1_bus_master1'

sandhan26, 2017-11-19:
@timofurrer I would like to close these issue, as i have confused between BME280 vs DS18xx sensors. 
I am extremely sorry for that.

bsimmo, 2017-11-19:
DS18B20 uses one-wire, this is not I2C. Hence you can ignore that the BME
sensor.

You do not need a 4K7 for the BME, but for the DS18B20 (between data and
3V3).

But I've just seen your last message, I guess you've realised :-)

For further help I suggest asking on the RaspberryPI forum itself.

timofurrer, 2017-11-19:
> I was logged into the default account 'pi'

You cannot load the kernel modules with the `pi` user. You have to be root. You can use `sudo` for that.

> For further help I suggest asking on the RaspberryPI forum itself.

I second that!

🍻

closed 2017-11-19 by timofurrer

RubensZimbres, 2018-06-27:
Hello Timo, I'm running a Python 3 code to collect temperature using a sensor (LM35) in a Raspberry Pi 3 running on Ubuntu Core snap classic, using w1thermsensor library.

I already edited the /boot/config.txt file as follows:

device_tree=bcm2708-rpi-b.dtb 
device_tree_overlay=overlays/w1-gpio-overlay.dtb
dtoverlay=w1-gpio,gpiopin=7
dtparam=i2c_arm=on

Also /etc/modules

w1-gpio
w1-therm

I changed the gpiopin from 4 to 7, but this does not seem to be an issue.

The problem is: when I run my Python script, I get the following error:

Traceback (most recent call last):
  File "Temp.py", line 2, in <module>
    from w1thermsensor import W1ThermSensor
  File "/home/rubenszmm/.local/lib/python2.7/site-packages/w1thermsensor/__init__.py", line 7, in <module>
    from .core import W1ThermSensor  # noqa
  File "/home/rubenszmm/.local/lib/python2.7/site-packages/w1thermsensor/core.py", line 316, in <module>
    load_kernel_modules()
  File "/home/rubenszmm/.local/lib/python2.7/site-packages/w1thermsensor/core.py", line 310, in load_kernel_modules
    raise KernelModuleLoadError()
w1thermsensor.errors.KernelModuleLoadError: Cannot load w1 therm kernel modules

This is the contents of /sys/bus/ folder. There is no w1 folder, so I can't run "modprobe w1-gpio"

amba         container     hid       mmc       sdio   usb
clockevents  cpu           i2c       platform  serio  workqueue
clocksource  event_source  mdio_bus  scsi      spi

Is there any fix I can do to overcome this error ?

Thanks in advance.

bsimmo, 2018-06-27:
That's not a 1-wire temperature sensor?

timofurrer, 2018-06-29:
> I'm running a Python 3 code
> ...
> Traceback (most recent call last):
> File "Temp.py", line 2, in
> from w1thermsensor import W1ThermSensor
> File "/home/rubenszmm/.local/lib/**python2.7**/site-packages/w1thermsensor/init.py", line 7, in 
> ...

It's not the issue here, but you are actually running Python 2.7 instead of Python 3.

> That's not a 1-wire temperature sensor?

Exactly - That's certainly one of the problems here.

However, you should be able to load the kernel modules any way. I guess you don't have the appropriate permissions. If you want to load the kernel modules from Python with `w1thermsensor` you have to run it as root. See https://github.com/timofurrer/w1thermsensor#disable-kernel-module-auto-loading

RubensZimbres, 2018-06-29:
Thanks @bsimmo and @timofurrer I will get the appropriate sensor.