timofurrer/w1thermsensor #45: water temp sensor - sleep() does not work as intended!!

Ricardosgeral, 2018-02-02:
I have a water temp sensor DS18B20 (https://www.adafruit.com/product/381)
It's connected to pin GPIO4 and it has a 10k resistor like recommended
I want to have readings every second
So I've done this 
`from time import sleep
from datetime import datetime
from w1thermsensor import W1ThermSensor

## Constants
MEASURE_INTERVAL = 1   #seconds

while True:
    d = datetime.now()
    time_ = '{:%H:%M:%S}'.format(d)
    # (DS18B) Water temperature sensor
    try:
        w = W1ThermSensor()
        water_temp = str(w.get_temperature())  # default pin GPIO4 for data and temp in Celsius degrees
    except:
        water_temp = '0'
    print(time_,water_temp)

    sleep(MEASURE_INTERVAL)`

However instead 1 sec interval I'm having 2 sec consistently (most of times)
this is the result

`17:20:55 19.437
17:20:57 19.437
17:20:59 19.437
17:21:01 19.437
17:21:03 19.437
17:21:05 19.437
17:21:06 19.437
17:21:08 19.437
17:21:10 19.437
17:21:12 19.437
17:21:14 19.437
17:21:16 19.437
17:21:17 19.437
17:21:19 19.437
17:21:21 19.437`

Why? I don't understand
should i change something in W1ThermSensor class?

bsimmo, 2018-02-02:
It takes about 1 second for the sensor to start, take, calculate and give
it's reading. (~750ms normally). It is not an instant reading and your
program will stop until the reading has been given.
Hence getting close to a 2second interval.

bsimmo, 2018-02-02:
And the 'mostly 2 seconds' is due to ~1.75 seconds being added up and eventually the rounding will look like 1 second instead of two second.
Add in milliseconds and you should see.

Ricardosgeral, 2018-02-05:
Ok thanks! SO my best option is to define

EFFECTIVE_INTERVAL = 1 #seconds
MEASURE_INTERVAL = EFFECTIVE_INTERVAL - 0.800   # to account with the delay of the sensor

this way it works 90% of times

timofurrer, 2018-02-05:
Thanks @bsimmo ! 
@Ricardosgeral does that solve your problem?

Ricardosgeral, 2018-02-05:
Yes! But I placed the reading in a separate Thread. Thanks

closed 2018-02-05 by timofurrer