timofurrer/w1thermsensor #48: Huge delay on CLI

Zach443, 2018-08-25:
I have a Pi 1, so I'm not sure if this is the cause, but when using the CLI there is a 10-20 second delay on every command. Is this normal? The command output is normal, but the delay seems insane.

bsimmo, 2018-08-26:
> Are you experiencing this delay when using the Python API, too? Or is it
> really just the CLI?
> Are you using sudo when running the CLI tool? If so, can you do a sudo su
> and verify that it's not sudo which causes that delay.
> If that's not the case, please set W1THERMSENSOR_NO_KERNEL_MODULE=1.
> Without this setting it'll always load the w1-gpio and w1-therm kernel
> modules when running the CLI.

I think I am seeing the delay too, but I had put it down to my setup.
(No sudo as don't need it).
I first noticed it a month or so ago iirc.

Just Pi Zero (I use an Google AIY Raspbian image, all up to date).
It take much longer than it used to.
I wonder if a kernel update has altered something.
I'll test with a fresh image later.

timofurrer, 2018-08-26:
hmmm, that's strange :thinking: 
Can you verify if the delay is coming from the `modprobe`?

Could you also do something like this:

```python
from time import time
from w1thermsensor import W1ThermSensor

sensor = W1ThermSensor()
print(time())
sensor.raw_sensor_value
print(time())
```

If you are seeing the delay with this, too the problem might be when open / reading the file.

Zach443, 2018-08-26:
For me, the delay is strictly with the CLI. Printing the reading to the console every second programmatically works just fine. The huge delay is only when using any operation with the CLI, whether that be a temp reading, listing available sensors, or anything else. I'm not totally sure by what you meant in your first response to do.

timofurrer, 2018-08-27:
Alright, please try:

`W1THERMSENSOR_NO_KERNEL_MODULE=1 w1thermsensor ls`

do you see the delay here, too?

bsimmo, 2018-08-27:
for me about 14 seconds (to read two sensors) with that, similar for the
normal method will all

RaspberryPi has all the gubbins loaded up at boot time (normally).

Zach443, 2018-08-31:
@timofurrer Same delay with that command.

timofurrer, 2018-09-02:
Could you please do:

```bash
sudo time ls -l /sys/bus/w1/devices
```

You'll see a directory for each sensor connected to your Pi. Once you spotted the right (maybe even the only) one, can you run the following: (whereas `<THE_SENSOR_DIRECTORY>` is the directory of your sensor)

```bash
sudo time cat /sys/bus/w1/devices/<THE_SENSOR_DIRECTORY>/w1_slave
```

Please post the output here, including the time infos.

timofurrer, 2018-11-01:
@Zach443 could you give it a try already?

bsimmo, 2018-11-01:
Just noticed I'm using 1.0.3 it seems
( python3-w1thermsensor/stable,now 1.0.3 all [installed] )
 which is the one in apt , so cannot use anything newer as that the default installed version.

Could you update apt so we can use 1.0.5 and see if that fixes it?

timofurrer, 2018-11-01:
Can you cat the file in the directory you've used? 
In `/sys/bus/w1/devices/28-031561e6d7ff` there should be a sensor file.

bsimmo, 2018-11-01:
Do you mean
```
pi@AIYTankPi:~ $ cat /sys/bus/w1/devices/28-031561e6d7ff
cat: /sys/bus/w1/devices/28-031561e6d7ff: Is a directory

pi@AIYTankPi:~ $ cd /sys/bus/w1/devices/28-031561e6d7ff
pi@AIYTankPi:/sys/bus/w1/devices/28-031561e6d7ff $ ls
driver  hwmon  id  name  power  subsystem  uevent  w1_slave

pi@AIYTankPi:/sys/bus/w1/devices/28-031561e6d7ff $ cat w1_slave
68 01 01 01 1f ff 0c 10 ac : crc=ac YES
68 01 01 01 1f ff 0c 10 ac t=22500

pi@AIYTankPi:/sys/bus/w1/devices/28-031561e6d7ff $ time cat w1_slave
68 01 01 01 1f ff 0c 10 ac : crc=ac YES
68 01 01 01 1f ff 0c 10 ac t=22500

real    0m0.908s
user    0m0.002s
sys     0m0.038s
```

timofurrer, 2018-11-02:
> Could you update apt so we can use 1.0.5 and see if that fixes it?

Unfortunately, I don't do this apt syncing. I'll ask for the update though!

timofurrer, 2018-11-02:
Seems pretty good. Can you try do the same from source, so you don't have to wait for apt to be updated with the newest release.

bsimmo, 2018-12-14:
Bump to remind me to check this (or the other chap @Zach443

timofurrer, 2019-03-08:
I'm going to close this, due to inactivity. 
@Zach443 when you are ready with the feedback - I can re-open. 🎉

closed 2019-03-08 by timofurrer