timofurrer/w1thermsensor #50: Unable to get Python 3 W1ThermSensor work with RPi 1-Wire Pi Plus adapter card

mikespolk, 2018-12-31:


mikespolk, 2018-12-31:
I think they already are the newest ones(?), as did the RPi scratch reinstall yesterday, and as listed from the Add/Remove Software:
(x) Python w1 therm sensor module (Python 3)
        python3-w1thermsensor-1.0.5
     Python w1 therm sensor module (Python 2)
        python-w1thermsensor-1.0.5

bsimmo, 2018-12-31:
> Do you get a reading not using the python software?
> i.e. hower the HAT says to get readings?

Forgot to say, I ask as I am assuming it is this hat
https://pinout.xyz/pinout/1_wire_pi_plus

You talk to a chip via i2c on the HAT, not with 1-wire directly, which this
module does.
See the link and the link on there to GitHub, their webpages etc.

mikespolk, 2018-12-31:
Yes, my 1-Wire adapter is just that formfactor that bsimmo sent the link for.
Also, to clarify, what already have been tried:

1. Just to make sure. to not interfere with any possible GPIO wirings I am going to use, I have disconnected any other connections from the GPIO pins.

2. With the (Python2 level) OWFS library as I mentioned above, the sensors enumerate OK, but not with the w1thermsensor library.

3. Did try many combinations in /boot/config.txt, /etc/modules, /etc/modprobe.d/raspi-blacklist.conf files, for the dtparam=i2c, dtparam=i2c1, i2c-bcm2708, i2c-bcm2835, i2c-dev, w1_gpio and w1_therm lines, to view if I have luck (=didn't have)

How to proceed next?

fireheadman, 2018-12-31:
How do I get off this list?  I keep unsubscribing then get another email.

bsimmo, 2018-12-31:
I guess mute (but I think that may just be the current issue) on the email
or on GitHub stop following/etc.
Have a look on GitHub, it'll be where you asked to follow.

mikespolk, 2019-01-01:
OK, let's move to Gitter, and then finally update status here, when ready.
Meanwhile, short answer: removed 1.0.5 with
 _sudo apt-get --purge remove python3-w1thermsensor_
and then installed 1.1.2 with
 _pip3 install w1thermsensor_
(hope, did that correctly, as aimed)
However, no sensors enumerated, yet.

bsimmo, 2019-01-01:
But your sensor HAT is i2c,
 1-wire is 1-wire.

You need to talk to the HAT with i2c.

This module doesn't do that.

timofurrer, 2019-01-01:
@bsimmo thanks for the answer, I think that's exactly the problem. I'm confused though as why the sensor appears in /sys/bus/w1/devices` with the `00` family code. 

Which is not supported by the w1 kernel module according to https://www.kernel.org/doc/Documentation/w1/slaves/w1_therm

bsimmo, 2019-01-02:
I think 00- could be 'ghost' sensors, i.e. not really there and/you have
bad wiring. I'd have to search on it.
Also no idea either.

mikespolk, 2019-01-02:
Just to mention, yesterday again I doublechecked the wirings by installing the owfs library to my from-scratch-installed RPi, according these articles: https://www.abelectronics.co.uk/kb/article/3/owfs-with-i2c-support-on-raspberry-pi and https://www.abelectronics.co.uk/kb/article/1/i2c--smbus-and-raspbian-linux, and my DS18B20 and DS18S20 sensors enumerated OK.

bsimmo, 2019-01-02:
So reading through that they are putting a 'driver' in to make fake 1-wire
files.
You'll need to modify this python module to pick up the specific serial
code it creates.
This module is not a driver, it just looks for valid serial numbers.
You'll still need to be running that owfs 'driver' also.

bsimmo, 2019-01-02:
for the info, I've been working on the emails, for some reason the initial post didn't send the actual post with it, so I just went off the title and assumed there was no description with it.
i'm on holiday and couldn't remember my GitHub login, I have now :-)

I understand the thread a bit better, same problem though.

bsimmo, 2019-01-02:
so had a read through.  I would suggest abondoning using this module for now.

look at the OWFS website and examples for reading the data from it the files or server/http port.
it's used a lot and there are many examples from a quick search (I have never used it but have seen posts on it on the RPi forums) 

but it look like it mounts the sensor in
/mnt/1wire
and reads from there as a file system, if you look there do you see them?
if yes then just redirect any python3 program to read from there rather than
/sys/bus/w1/devices/

then you'll not need to use the python server module.
you will still need to have owfs running as that is taking the i2c reading from the chip on the hat and doing the conversion.

an alternative method is to see how to talk directly to the chip on the hat via i2c to get it's readings, but that sounds more complicated ;-)

mikespolk, 2019-01-02:
Yes, have to try that. Essentially, in that configuration, the owfs must be defined to autostart at boot, and then let it generate the temperature readings to /mnt/1wire folder, asynchronously in the background. Then my Py3 app would read the file contents when needed.

For that, I think that these articles would serve best:
https://raspberrypi.stackexchange.com/questions/62292/usage-of-python-ow-one-wire-file-system-python-package-for-reading-1-wire-ds18b2
https://wiki.m.nu/index.php/OWFS_with_i2c_support_on_Raspberry_Pi_(English_version)
because the autostart definition is there, too.

What I have to do first is to again create a fresh boot Raspbian image from scratch, to prevent existing w1thermsensor and its definitions or any other trial+error configurations from causing any secondary problems. Will take its time, and I will provide update then.

mikespolk, 2019-01-03:
Yes, I got owfs background processes working, by following the above articles, essentially this: https://wiki.m.nu/index.php/OWFS_with_i2c_support_on_Raspberry_Pi_(English_version)
- 1-Wire sensors are connected via the HAT type 1-Wire Pi Plus adapter card
- owfs autostarts at boot
- In /mnt/1wire folder, there starts appearing 10* and 28* subfolders, for  DS18S20 and DS18B20 sensor types, respectively
- The up-to-date temperatures are readable in the temperature text files.
So I can have my Py3 application, running in the foreground, to read temperatures from text files.
The temperature readings seem to update automatically every 10 seconds or so.

Thank you timofurrer and bsimmo for Support and troubleshooting!
I would however be very glad, if the w1thermsensor module would support the i2c bus, too, someday.
That is because in home automation, where the cabling may be lengthy, 1-Wire adapter car protects RPi internal bus from electrical interference, and would let it run more stable from day to day.

bsimmo, 2019-01-03:
In linux everything is a text file, the 1-wire is a text file as such too.
Just done at slower level in the OS (the bits you load at the start up)
If it looks the same owfs mnt folder then exerting DS18B20s code should
work but pointed the the different directory.
Try what I mentioned above.

I personally don't think this module should support i2c reading. At the
moment this module just passed a text file.
The OS is doing what OWFS is doing but for one wire at an OS module level.
Leave it to OWFS to do the low level stuff, it designed for it and used in
industry. Keep the expertise in one place.
Or get OS to do it at the OS level.

timofurrer, 2019-01-04:
Thanks @bsimmo for the support!

> I personally don't think this module should support i2c reading

What's great about this module is it's simplicity - in maintenance and usage. Therefore I want add i2c support any time soon - except someone has a really good point way it should be implemented.

> Try what I mentioned above.

@mikespolk I'd be interested about that, too.
Can you post the output of `ls -lR /mnt/1wire` here? 

If that's working I'll add an option to configure this path.

mikespolk, 2019-01-04:
Attached is listing you asked for, from a booted, idle RPi, having those owfs autostart stuff installed.
[mnt-1wire-folder-listing.txt](https://github.com/timofurrer/w1thermsensor/files/2727388/mnt-1wire-folder-listing.txt)

mikespolk, 2019-01-04:
After all of the previous discussions, and because I would love to see things trivial where possible, I am quite confused with one thing: 
Why wouldn't there just be a module that, as its only task, executes a 1-wire read sequence thru the i2c bus, and then synchronously returns the results to the calling Py application? Instead of acting as a background server and using a drop-in directory structure to pass results?
Or, is this question irrelevant in some way?

bsimmo, 2019-01-04:
There may be one, I've never used an i2c module for 1-wire.
If you know the i2c specs from the chipset datasheets you could probably
make one.
Grab the chip name and search on it.

closed 2019-01-06 by mikespolk

MaherKhaz, 2023-01-14:
l have the same problem. when l run  w1thermsensor ls, there is no sensors found. The ds18b20 is emitting a red light. Qhat does it mean. It is plugged to pin 8 or ceo.