timofurrer/w1thermsensor #52: async support

thijstriemstra, 2019-01-17:
Would be nice to use the async support in python 3.5/3.6.

timofurrer, 2019-01-19:
Good idea! I'll have some time to look into it end of the month.

buldre, 2019-10-01:
Will this enable higher logging frequency with reduced resolution/precision?
Currently I managed to set precision from 12 to 10 bits in the RPI Termnial, but the 12 bits interval of 750 ms (+overhead) between reads remained unchanged. I expected to see 4 times higher frequency.
Whether reading the same sensor, or 2 sensors, the reading interval is the same. I updated the Python 3 library yesterday.

timofurrer, 2019-11-03:
> Will this enable higher logging frequency with reduced resolution/precision?

No, this has nothing to do with the precision or resolution of the sensor itself. 

> I expected to see 4 times higher frequency.

I think, we've clarified this via email, right? If not, please open a separate issue :tada:

buldre, 2019-11-03:
Hi Timo!

You explained this very well!

Thanks,
Ben Velde

jonte, 2020-02-06:
It seems like asyncio[0] doesn't support asynchronous file I/O (for good reasons)[1].

One way to still provide an interface which can be consumed by asyncio-based python code would be to execute the blocking read[2] in a thread pool. There's support for this in asyncio, using `loop.run_in_executor`[3]. It would look something like this:

```python
    async def raw_sensor_strings_async(self):
        try:
            with open(self.sensorpath, "r") as f:
                with concurrent.futures.ThreadPoolExecutor() as pool:
                    data = await loop.run_in_executor(pool, f.readlines)
        except IOError:
            ....
```

In my current code, which uses the W1ThermSensor library, I am wrapping the call to `get_temperature` in a similar fashion to the example above.

Given that we can't make the file I/O truly asynchronous, does it make sense to provide an async helper such as the one seen above (of course, we would have to make all the calls out to, and including `get_temperature` async as well)?

I guess it would be convenient to not have to wrap `get_temperature` yourself, but I am unsure if it makes sense to implement this. It might not be obvious to the user of such an async method that threads are used, and might to more harm than good. What do you think?

I could take a stab at a PR if we still want this functionality.

[0]: https://docs.python.org/3/library/asyncio.html
[1]: https://github.com/python/asyncio/wiki/ThirdParty#filesystem
[2]: https://github.com/timofurrer/w1thermsensor/blob/master/w1thermsensor/core.py#L235
[3]: https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.loop.run_in_executor

commit e6f99ed9 (2020-06-20): Add asyncio support with AsyncW1ThermSensor. Closes #52, #67
commit 3332f4b4 (2020-06-20): Add asyncio support with AsyncW1ThermSensor. Closes #52, #67
commit 9e36b7fa (2020-06-20): Add asyncio support with AsyncW1ThermSensor. Closes #52, #67
closed 2020-06-20 by timofurrer