timofurrer/w1thermsensor #54: Code stopped by SensorNotReady Error

miqi95, 2019-04-23:
First of all, great library, it's saved me a lot of time setting up my temperature monitoring system. I'm running this code to read from 12 different DS18B20 sensors every 30 secs and write to a CSV. The first time I ran it, it read about 210 times before throwing this error. Last time, only 2 reads before stopping. What is the problem here?
`from w1thermsensor import W1ThermSensor
import csv
import time

#Sensors are labeled according to Shelf-Location, e.g. TL=Top-left.
#Left/Right based on facing into fridge from door

sensorTF=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000b20a912")
sensorTB=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000b208271")
sensorTR=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000a9ae2c7")
sensorTL=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000b1fde3a")
sensorMF=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000b201718")
sensorMB=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000a99ff32")
sensorMR=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000b2065d1")
sensorML=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000b1ffd03")
sensorBFR=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000a9a1729")
sensorBFL=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000a9a972b")
sensorBBR=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000a99b35a")
sensorBBL=W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000b1fcb31")

with open('IceboxTemps.csv', 'w') as f:
	thewriter = csv.writer(f)

	thewriter.writerow(['Time (s)', 'TopFront', 'TopBack', 'TopRight', 'TopLeft', 'MidFront', 'MidBack', 'MidRight', 'MidLeft', 'BotFrontRight', 'BotFrontLeft', 'BotBackRight', 'BotBackLeft'])
	for i in range(1,721):
		t=(i-1)*30
		thewriter.writerow([t, sensorTF.get_temperature(), sensorTB.get_temperature(), sensorTR.get_temperature(), sensorTL.get_temperature(), sensorMF.get_temperature(), sensorMB.get_temperature(), sensorMR.get_temperature(), sensorML.get_temperature(), sensorBFR.get_temperature(), sensorBFL.get_temperature(), sensorBBR.get_temperature(), sensorBBL.get_temperature()])
		time.sleep(30)

 `

timofurrer, 2019-04-24:
That happens only in a [single case](https://github.com/timofurrer/w1thermsensor/blob/master/w1thermsensor/core.py#L216): if the w1-therm kernel module reports that the sensor is not ready to read. From the [kernel module specs](https://www.kernel.org/doc/Documentation/w1/slaves/w1_therm):

> The first line contains the nine hex bytes read along with a calculated crc value and YES or NO if it matched.

My bet would be that there is something wrong with the wiring, like loose connections or something, so that the CRC doesn't match.

As far as I can tell it's not an issue of this module, so I'm going to close this issue.

miqi95, 2019-04-24:
If there was a loose connection, how could each sensor provide a measurement even some of the time? There aren't any vibrations in my system that could mess with the CRC while taking measurements. How long does it take a sensor to be ready to read? I'm not sure I fully understand this process. Could I be using wires that are too long and my signal is weak?

bsimmo, 2019-04-24:
I don't know you setup, so will have to talk generically.
Each sensor will, if set to defaults, about 750ms and will block the code
until it completes.
It sends a request, the sensor then gets on with it with it (takes lots of
reading and builds up the measurement), then send the result back.
Once the is done, your code will then move on to the next sensor.

The data line need to be above a certain voltage at the sensor to generate
the 'initial read request' (if I remember it correctly, it on the spec
sheet).