timofurrer/w1thermsensor #55: Unable to locate package python3-w1thermsensor

natanbenchimol, 2019-06-18:
Hey,

I've used this package before and really loved how easy it was to use.
I am having issues downloading the package this time. When I try to install it on Rasbian using
sudo apt-get install python3-w1thermsensor
I get the error: "_unable to locate package python3-w1thermsensor_"
I am using a raspberry pi with rasbian.
Any chance you can help me out?

timofurrer, 2019-06-18:
What version of raspbian are you using?
Did you do an `apt update` ?

timofurrer, 2019-06-18:
The raspbian apt repositories contain the `w1thermsensor` packages. See http://archive.raspberrypi.org/debian/dists/stretch/main/binary-armhf/Packages

Make sure you have an up-to-date system (`apt update`).

I'm going to close this for now. You can still post to it if the problem persists and we go from there. 🎉

closed 2019-06-18 by timofurrer

ComputerSmiths, 2022-09-08:
I'm seeing this in the Stretch packages above, and in Buster but not Bullseye, how do I install it?

Thanks!

bsimmo, 2022-09-08:
As it says on the front page (and for most python modules)

pip install w1thermsensor

ComputerSmiths, 2022-09-08:
Thanks, pip3 install worked, though apt-get used to.