timofurrer/w1thermsensor #57: Citation

Gbassous, 2019-07-24:
I used your library in my dissertation, can you confirm my citation?

T. Furrer, “w1thermsensor” , 2019. [Online]. Available: https://github.com/timofurrer/w1thermsensor. [Accessed: 24-Jul-2019].

Since there are several contributors without name or contact info, if you'd like me to reference them all, could you provide some information on them, please?

timofurrer, 2019-07-25:
> Since there are several contributors without name or contact info, if you'd like me to reference them all, could you provide some information on them, please?

Even if I'd like you to reference them with some information, like the name, it probably wouldn't be possible. So for me your suggested citation is absolutely fine.

Out of curiosity is this a publicly accessible dissertation?

closed 2019-07-25 by timofurrer

jamesmyatt, 2019-07-25:
If you're doing any data analysis, then also see https://the-turing-way.netlify.com/introduction/introduction

Gbassous, 2019-07-25:
It will be publicly available, once it's done, I'll link it here. There will also be one or two papers derived from it, I'll provide them too.

@jamesmyatt That's really cool and important. I will look in that guide once i finish it and definitely will make data and code available. I should put my code in the annexes of the dissertation and I'll provide a link to whomever want's to download the data. At the moment  I'm just rushing to wrap things up due to very close deadlines.