timofurrer/w1thermsensor #65: Wiring Diagram suggestion

jesse1d, 2019-11-25:
The wiring diagram appears to show a pull down resistor, connecting the signal line to ground. This should be, IMHO, a pull-up resistor connected to the supply line.
...but thanks for publishing a brilliant library that makes a very complex interface really easy to implement.
Dave

t7ko, 2019-12-01:
Yes, now that I looked at it, all other resources show it as pull up...  And I have already soldered and packaged it the other way following the diagram :(

apocalipsys, 2019-12-09:
> The wiring diagram appears to show a pull down resistor, connecting the signal line to ground. This should be, IMHO, a pull-up resistor connected to the supply line.
> ...but thanks for publishing a brilliant library that makes a very complex interface really easy to implement.
> Dave
so, we have to connect like this?
Raspi VCC (3V3) Pin 1 -----------------------------   VCC    DS18B20
                                               |
                                               |
                                               R1 = 4k7 ...10k 
                                               |
                                               |
Raspi GPIO 4    Pin 7 -----------------------------   Data   DS18B20

Raspi GND       Pin 6 -----------------------------   GND    DS18B20

timofurrer, 2019-12-11:
The README should be fixed now. Thanks for reporting and sorry for the troubles it may has caused you. :tada:

commit 227a9df8 (2019-12-11): Corrected the placement of the resistor. Pull Up not pull down. Closes #65 (#66)
closed 2019-12-11 by timofurrer