timofurrer/w1thermsensor #67: Is a nonblocking interface possible?

muratisik-lab, 2019-12-29:
I am testing the package on my RPi 3 for use in a home automation system, and so far it works fine.

The only problem I have is that every call to sensor.get_temperature() blocks my program for about 1 second. If only we had a non-blocking interface to utilize the 1-second for other useful tasks.

Something like the following:

        sensor.start_conversion()
        #do other stuff here
        if sensor.is_conversion_ready():
                temp=sensor.get_temperature()

Many thanks

Murat Isik
murat . isik . istanbul @ gmail

bsimmo, 2019-12-29:
Threading was suggested some time ago but not rejected.  You are free to
implement it yourself.

muratisik-lab, 2019-12-30:
Hi Timo,

Below is a simple non-blocking implementation based on threading.
If you are interested, I can reinforce and contribute it to your package.

Happy new year

import threading
from w1thermsensor import W1ThermSensor
from time import sleep

class SensorMonitor():
	class MonitorThread(threading.Thread):
		def __init__(self, s, m):
			super().__init__()
			self.__sensor=s
			self.temperature=None
			self.__mutex=m
			self.__Continue_Monitoring=True

		def StopMonitoring(self):
			self.__Continue_Monitoring=False

		def run(self):
			while self.__Continue_Monitoring:
				self.__t=self.__sensor.get_temperature()
				self.__mutex.acquire()
				self.temperature=self.__t
				self.__mutex.release()

	def __init__(self, s):
		self.__sensor=s
		self.__temp_mutex=threading.Lock()
		self.__Continue_Monitoring=False
		return

	def Start(self):
		self.__w=self.MonitorThread(sensor, self.__temp_mutex)
		self.__w.daemon=True
		self.__w.start()
		return

	def Stop(self):
		self.__w.StopMonitoring()
		self.__w.join()
		return

	def ReadTemperature(self):
		self.__temp_mutex.acquire()
		self.__read_temp=self.__w.temperature
		self.__temp_mutex.release()
		return(self.__read_temp)

if __name__ == "__main__":
	Monitor={}

	ListOfSensors=W1ThermSensor.get_available_sensors()

	for sensor in ListOfSensors:
		Monitor[sensor]=SensorMonitor(sensor)
		Monitor[sensor].Start()

	for i in range(25):
		print("Iteration: "+str(i))
		for sensor in ListOfSensors:
			print(sensor.id+": ", Monitor[sensor].ReadTemperature())
		sleep(0.1)

	for sensor in ListOfSensors:
		Monitor[sensor].Stop()

Iteration: 0
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 1
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 2
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 3
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 4
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 5
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 6
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 7
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 8
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 9
000003de7e98:  None
000003de3d49:  None
000003de3c00:  None
Iteration: 10
000003de7e98:  23.062
000003de3d49:  None
000003de3c00:  None
Iteration: 11
000003de7e98:  23.062
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 12
000003de7e98:  23.062
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 13
000003de7e98:  23.062
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 14
000003de7e98:  23.062
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 15
000003de7e98:  23.062
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 16
000003de7e98:  23.062
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 17
000003de7e98:  23.062
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 18
000003de7e98:  23.062
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 19
000003de7e98:  23.125
000003de3d49:  23.187
000003de3c00:  23.0
Iteration: 20
000003de7e98:  23.125
000003de3d49:  23.25
000003de3c00:  23.0
Iteration: 21
000003de7e98:  23.125
000003de3d49:  23.25
000003de3c00:  23.062
Iteration: 22
000003de7e98:  23.125
000003de3d49:  23.25
000003de3c00:  23.062
Iteration: 23
000003de7e98:  23.125
000003de3d49:  23.25
000003de3c00:  23.062
Iteration: 24
000003de7e98:  23.125
000003de3d49:  23.25
000003de3c00:  23.062

timofurrer, 2020-06-20:
The current master branch and the upcoming 2.0.0 release will feature a async interface with `AsyncW1ThermSensor`.
There is still some testing to be done - See the README for more information ;)

closed 2020-06-20 by timofurrer

muratisik-lab, 2020-06-20:
Wow great news. Thank you Timo.