timofurrer/w1thermsensor #73: Changing resolution DS18B20 doesn't change speed

PieterHerings, 2020-03-05:
Hi,

For a master's project I need to read two DS18B20 sensors for 30 seconds as much as possible. The 12 bit resolution takes ~750ms, so with two sensors it takes 1.5 sec for a measurement. To be quicker I changed the resolution to 11 bit. The accuracy of the sensor went down, but the speed of the readings doesn't improve. I can see on the specifications of the DS18B20 that the speed of a 11 bit reading should be ~375 ms instead of ~750 ms. 

Why doesn't the sensor speed go up? Do you have a tip to get the sensor readings quicker?
I am using w1thermsensor 0.3.2 and to get a reading I just try 'sudo w1thermsensor all' in the terminal.

Thanks in advance

bsimmo, 2020-03-05:
I can't say why as I don't know.

But you also need to look at some sort of threading and asking for the two
readings one after the other and then reading the response.
You'll get both results in a second or less.

Much harder to code though, I had some code but no idea where I out it.

timofurrer, 2020-06-24:
w1thermsensor 2.0 (current master) provide an `AsyncW1ThermSensor` class which supports the async / await interface to read temperature. I've added some examples here: https://github.com/timofurrer/w1thermsensor/tree/master/examples

You could also just use whatever release you have right now and read the temperatures from two different threads (keep in mind that Python has a GIL)

Generally, `w1thermsensor` has little overhead to read temperatures - also keep in mind that Python is rather slow and therefore probably not suited for real-time applications - thus, the speed is mostly dominated by the time it takes to read the sensor files from the kernel modules.

closed 2020-06-24 by timofurrer