timofurrer/w1thermsensor #75: get_temperature does not raise raw_sensor_count exception

riban-bw, 2020-04-06:
Calling *get_temperature* never gives ResetValueError.
 
Within the method *get_temperature* there is a [check](https://github.com/timofurrer/w1thermsensor/blob/c181bd01112b6a02423d3ca5a5d56f12bf76975f/w1thermsensor/core.py#L334) whether the sensor is in reset phase which should raise a ResetValueError however the check is made on the raw data rather than the refactored data hence is never 85.0. The check probably be done on the value returned by *raw_sensor_count* with appropriate value.

timofurrer, 2020-06-24:
I cannot reproduce this issue. When I unplug the power my sensors yield a `w1_slave` file content along the lines of:

```python
50 05 4b 46 7f ff 0c 10 1c : crc=1c YES
50 05 4b 46 7f ff 0c 10 1c t=85000
```

This correctly results in a raised `ResetValueError` for the sensors listed here: https://github.com/timofurrer/w1thermsensor/blob/5fcf98ce1b8c1107edb8faa8f0d0f9c2561f263b/src/w1thermsensor/sensors.py#L23

What sensor are you using?

timofurrer, 2020-08-24:
Let me know if you have an update.

closed 2020-08-24 by timofurrer