timofurrer/w1thermsensor #89: Need to have separate GPIO pins for Tx and Rx

mikespolk, 2020-06-22:
Currently, the 1-wire devices can easily be connected to single GPIO line, like GPIO4.
That kind of usage is very demonstrative, because it is easy to connect one or more DS18B20's there and start measuring. 
However, there is need connect sensors via optocoupler, which help in protecting RPi devices from electrical glitches, especially from long distance wiring.
But connecting optocouplers is fairly difficult, in such bi-directional input/output channel usage, which 1-wire uses.

Resolution: Imagine if it was possible to define separate GPIO pins for Tx and Rx, for example with
  dtoverlay=w1-gpio,gpiotx=4,gpiorx=17
Then there would be two lines from RPi, and wiring would be greatly easier.

Can the w1thermsensor already be used that way, or is it possible to modify it like that?

bsimmo, 2020-06-22:
This is just a python module to use what Linux provides along with the OS. But I don’t think that is possible as that is not how 1-wire works.  You would need to talk to the kernel developers or try a bitbang method, still, how would you make onewire input act as two wire, how would 1 -wire know which wire to send it down.  I think you are looking at this all wrong.  Look up 1-wire and it different setups.  Unless you are thinking of the 1-wire/uart setups  Iknow little about.  That a Linux/os problem, the module just read a ‘file’ the OS makes.Ben Sent from Mail for Windows 10 From: mikespolkSent: 22 June 2020 16:22To: timofurrer/w1thermsensorCc: SubscribedSubject: [timofurrer/w1thermsensor] Need to have separate GPIO pins for Tx and Rx (#89) Currently, the 1-wire devices can easily be connected to single GPIO line, like GPIO4.That kind of usage is very demonstrative, because it is easy to connect one or more DS18B20's there and start measuring.However, there is need connect sensors via optocoupler, which help in protecting RPi devices from electrical glitches, especially from long distance wiring.But connecting optocouplers is fairly difficult, in such bi-directional input/output channel usage, which 1-wire uses.Resolution: Imagine if it was possible to define separate GPIO pins for Tx and Rx, for example withdtoverlay=w1-gpio,gpiotx=4,gpiorx=17Then there would be two lines from RPi, and wiring would be greatly easier.Can the w1thermsensor already be used that way, or is it possible to modify it like that?

mikespolk, 2020-06-22:
Thanks bsimmo for replying.
In response to your question; please find attached picture of my idea. With two GPIO lines, the first permanently as OUT and the other as IN types, they would work together to communicate with single 1-wire sensor.
[image]
I would imagine that the library routine would be even slightly simpler, because it does need to switch the single GPIO line between the OUT and IN modes. Instead, the routine would send Tx in the first GPIO and receive Rx from the second one.

Any idea, how that would be possible?

timofurrer, 2020-06-23:
As @bsimmo already mentioned: this w1thermsensor package just makes use of whatever the Linux kernel modules provide. Your idea isn't supported and you'd really have to talk to the kernel developers :)

Therefore, I am going to close this issue - if you like to keep discussing this here - feel free to do so.

Thanks any ways!

closed 2020-06-23 by timofurrer

mikespolk, 2020-06-23:
OK, thank you, that makes sense.

Just a silly question from total linux noobie;

What would "talk to the kernel developers :)" mean? Would that really be relevant idea or not? If yes, what would be the right way for that?

timofurrer, 2020-06-24:
hmm, not sure if you'll have success - as I imagine it's pretty hard to get changes in there. 
Maybe the author named [here](https://www.kernel.org/doc/Documentation/w1/slaves/w1_therm) is still active?