timofurrer/w1thermsensor #94: IndexError: list index out of range

bggardner, 2020-08-18:
I have four DS18B20 sensors on the 1-wire network of a Raspberry Pi.  After a long time (multiple hours, actual time may vary) of calling `get_temperature()` every 30 seconds in a while loop, I get an `IndexError` with the following trace:
```
Aug 18 03:19:00 hot-tub balboa-mqtt.py[9327]:   File "/usr/local/lib/python3.7/dist-packages/w1thermsensor/core.py", line 263, in get_temperature
Aug 18 03:19:00 hot-tub balboa-mqtt.py[9327]:     raw_temperature_line = self.get_raw_sensor_strings()[1]
Aug 18 03:19:00 hot-tub balboa-mqtt.py[9327]:   File "/usr/local/lib/python3.7/dist-packages/w1thermsensor/core.py", line 245, in get_raw_sensor_strings
Aug 18 03:19:00 hot-tub balboa-mqtt.py[9327]:     if data[0].strip()[-3:] != "YES" or "00 00 00 00 00 00 00 00 00" in data[0]:
Aug 18 03:19:00 hot-tub balboa-mqtt.py[9327]: IndexError: list index out of range
```
After I restart the script, everything is happy, but it eventually occurs again.  I will be adding some logging and updating this issue as I discover more.  Any guidance is appreciated.

**Environment and Version**
* Hardware: Raspberry Pi 2 Model B Rev 1.1
* OS: Raspbian OS Buster 10.4 (latest updates)
* Python Version: 3.7.3
* w1thermsensor Version: 2.0.0a1 (issue also exists with distro release)

timofurrer, 2020-08-18:
Do you mind changing the source a little for your installation - so that `data[0].strip()` and the `[-3:]` is done on different lines and log the individual results? Like the contents of `data`, `data[0].strip()` ?

bggardner, 2020-08-18:
I added some logging to print `data` and its length.  I'll post the results, but it will probably take a while.  I suspect `data` is an empty string...maybe the read occurs right when the OS is trying to write to the file?

timofurrer, 2020-08-18:
> maybe the read occurs right when the OS is trying to write to the file?

I'm not really sure how atomic that operation is on `/sys`. Might be the case though. 
Or maybe a defective connection or something similar could cause this.

closed 2020-08-24 by timofurrer

tomneko, 2021-12-16:
Hi I use Python try..except block. It is works fine with ver.1.0.5

#!/usr/bin/python3
from w1thermsensor import W1ThermSensor
import time

sensor = W1ThermSensor()

while True:
        try:
                temperature_in_celsius    = sensor.get_temperature(W1ThermSensor.DEGREES_C)
                print("celsius:    {0:.3f}".format(temperature_in_celsius))
        except IndexError as e:
                print(e)
                print(type(e))
        # Error
        finally:
                time.sleep(5.0)

commit 0d60feaf (2020-08-24): Merge pull request #95 from bggardner/master

Resolve #94