timofurrer/w1thermsensor #96: Received IndexError instead of SensorNotReadyError from get_raw_sensor_strings() when reading DS18B20

vollbier, 2020-08-30:
I am having a few DS18B20 sensors connected to raspberry pi zero. From time to time the contents of the file ``/sys/bus/w1/devices/28.../w1_slave`` appears to be empty (not sure why). Hence the contents of the list ``data`` in the method ``get_raw_sensor_strings()`` in file ``async_core.py`` is empty as well. Accessing it raises an ``IndexError``.

Changing line 65 in file ``async_core.py`` from:
```
     if (
            data[0].strip()[-3:] != "YES" or "00 00 00 00 00 00 00 00 00" in data[0]
        ):  # pragma: no cover
            raise SensorNotReadyError(self) 
```
to
```
     if (
           not data or data[0].strip()[-3:] != "YES" or "00 00 00 00 00 00 00 00 00" in data[0]
        ):  # pragma: no cover
            raise SensorNotReadyError(self) 
```
resolves the problem.

Thanks.

PS: I am using version ``w1thermsensor 2.0.0a1``

timofurrer, 2020-08-31:
Thanks for the report. We've already fixed that for the non-async version of the code, see #94 . 

I've made the same change now in the async code and will release asap.

timofurrer, 2020-09-13:
I've released v2.0.0a2

closed 2020-09-13 by timofurrer

flyte, 2021-01-24:
Hey, any chance this bugfix could be backported to the 1.3.x version? Some of our Python 2 using users at flyte/pi-mqtt-gpio are hitting this issue.

bsimmo, 2021-01-24:
Or they could fork it and make a python2 port.

or get with the game and update to python3 ;-)

flyte, 2021-01-24:
Just hoped it'd be a trivial fix.

We're only supporting Python 2 until it becomes too much of a pain in the butt, or I finally release the Python 3 only asyncio version.

timofurrer, 2021-01-24:
To be honest, the fix would be trivial - however, the very last Python 2 release for w1thermsensor already happened and since it's long been deprecated any ways I don't want to support it any longer.

If you really need Python 2 I suggest that you go with @bsimmo solution of a fork-and-port. 

In any case, I highly recommend you to drop Python 2 support too and go with a supported Python 3 version - but I suspect you are aware of that already :)

🍻