timofurrer/w1thermsensor #97: ImportError: cannot import name Sensor

linusmmueller, 2020-09-13:
Hello,

I've just installed the code according to the guide and without any problems on an updated Raspian and the program manages to read both of my sensors, but when I try to use the code to change the resolution, I receive this error:

Traceback (most recent call last):
  File "Test.py", line 2, in <module>
    from w1thermsensor import W1ThermSensor, Sensor
ImportError: cannot import name Sensor

This is my code:

from w1thermsensor import W1ThermSensor, Sensor

sensor = W1ThermSensor(Sensor.DS18B20, "0301a279fa6e")
sensor.set_resolution(9)

Thanks in advance for your help.

timofurrer, 2020-09-13:
Which version of w1thermsensor do you have installed?
It's an alpha version (use `pip install --pre`) and not available as debian package.

linusmmueller, 2020-09-14:
Thanks for your help, I was running v 1.3.0 and after I uninstalled the version and installed version 2.0.0a2 it worked.

closed 2020-09-14 by linusmmueller