timofurrer/w1thermsensor #98: Intermittent Connection Crashes W1

David-Enst, 2020-10-24:
I updated my Pi hardware and BCRobotics Weather HAT.  It turns out that I had an intermittent connection problem with my temperature sensor. WeeWX (Python V3) would run for hours and then crash (see error below).

It would be nice to have to W1 code just log the error but carry on and not terminate.

---------- Error -----------------
Oct 14 05:00:17 IOTTAWA98 weewx[12418] INFO weewx.manager: Added record 2020-10-14 05:00:00 EDT (1602666000) to database 'weewx'
Oct 14 05:00:17 IOTTAWA98 weewx[12418] INFO weewx.manager: Added record 2020-10-14 05:00:00 EDT (1602666000) to daily summary in 'weewx'
Oct 14 05:00:17 IOTTAWA98 weewx[12418] INFO weewx.restx: Wunderground-PWS: Published record 2020-10-14 05:00:00 EDT (1602666000)
Oct 14 05:00:19 IOTTAWA98 weewx[12418] INFO weewx.engine: Main loop exiting. Shutting engine down.
Oct 14 05:00:19 IOTTAWA98 weewx[12418] INFO weewx.engine: Shutting down StdReport thread
Oct 14 05:00:22 IOTTAWA98 weewx[12418] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 5.26 seconds
Oct 14 05:00:27 IOTTAWA98 systemd[1]: apt-daily.service: Succeeded.
Oct 14 05:00:27 IOTTAWA98 systemd[1]: Started Daily apt download activities.
Oct 14 05:00:37 IOTTAWA98 weewx[12418] INFO weewx.imagegenerator: Generated 30 images for report SeasonsReport in 14.59 seconds
Oct 14 05:00:37 IOTTAWA98 weewx[12418] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__: Caught unrecoverable exception:
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****  list index out of range
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****  Traceback (most recent call last):
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****    File "/usr/share/weewx/weewxd", line 154, in main
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****      engine.run()
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 188, in run
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****      for packet in self.console.genLoopPackets():
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****    File "/usr/share/weewx/user/BCRobotics.py", line 229, in genLoopPackets
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****      readings = StationData.get_readings()
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****    File "/usr/share/weewx/user/BCRobotics.py", line 301, in get_readings
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****      if TempSensor : out_Temp = ds18b20.get_temperature()
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****    File "/usr/lib/python3/dist-packages/w1thermsensor/core.py", line 230, in get_temperature
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****      return factor(self.raw_sensor_value)
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****    File "/usr/lib/python3/dist-packages/w1thermsensor/core.py", line 193, in raw_sensor_value
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****      if data[0].strip()[-3:] != "YES":
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****  IndexError: list index out of range
Oct 14 05:00:37 IOTTAWA98 weewx[12418] CRITICAL __main__:     ****  Exiting.
Oct 14 05:04:39 IOTTAWA98 rngd[370]: stats: bits received from HRNG source: 2120064
Oct 14 05:04:39 IOTTAWA98 rngd[370]: stats: bits sent to kernel pool: 2073312
Oct 14 05:04:39 IOTTAWA98 rngd[370]: stats: entropy added to kernel pool: 2073312
Oct 14 05:04:39 IOTTAWA98 rngd[370]: stats: FIPS 140-2 successes: 106

bsimmo, 2020-10-24:
First update your python3 module, it may well be an old one.
 sudo pip3 install w1thermsensor -U
(assuming you used sudo before to install it)
use
pip3 freeze
first to check the current version.

I was still at 1.1.2.
there is also a newer version, but it's labelled in testing (v2...)

If you used apt to install w1thermsensor, then remove it and use pip3,
since that version has not been updated in a long time and people are still
using 1.0.5 using that method. :-(

I remember some out of bounds work being implemented, but that might be in
v2 ?

Ben

timofurrer, 2020-10-29:
> I remember some out of bounds work being implemented, but that might be in

Yes, that's true. Please use this version from pypi for now: https://pypi.org/project/w1thermsensor/2.0.0a2/

You might also just catch the thrown `IndexError` by yourself and let your service running ;)

closed 2020-10-29 by timofurrer