timofurrer/w1thermsensor #101: ds18b20 vs w1thermsensor

wowonk, 2021-01-09:
Hi,
sorry I'm only a Dummy.
But it seems to me, that the given examples do not work, because it is only possible to import the W1ThermSensor Class.
Sorry, if I wrong.
Gruss, wowonk

timofurrer, 2021-01-09:
Please use the issue template when reporting a bug: https://github.com/timofurrer/w1thermsensor/issues/new?assignees=&labels=&template=bug_report.md&title=

We can't help you without e.g. knowing which version of the package you are using  or which example you are referring to ...

closed 2021-01-09 by timofurrer

wowonk, 2021-01-10:
Sorry, to complicated for me.
--
                             Gruss, wowonk

bsimmo, 2021-01-10:
Basically it needs more detail.

Post you program using it

Tell us what you are running it on, P
The device, the operating system, at least give some sort of clue.

Tell us the error python3 gives.

Or its just guessing and you'll have to work it out yourself.