timofurrer/w1thermsensor #103: ImportError: cannot import name 'Sensor' from 'w1thermsensor'

a-sp, 2021-01-29:
**Describe the bug**
A clear and concise description of what the bug is.

Hello, when I try to run a simple code (see code below) I receive this error: 
```
Traceback (most recent call last):
  File "temperature.py", line 1, in <module>
    from w1thermsensor import W1ThermSensor, Sensor
ImportError: cannot import name 'Sensor' from 'w1thermsensor' (/usr/lib/python3/dist-packages/w1thermsensor/__init__.py)
```

**Environment and Version**
* OS (incl. distribution and version): Raspbian (10 Buster)
* Python Version:  3.7.3
* w1thermsensor Version (`w1thermsensor --version`):  2.0.0
* If applicable: DS18B20 wired to a Raspberry Pi 3 B+

**To Reproduce**
 w1thermsensor installed via `sudo apt-get install python3-w1thermsensor`

simple test code:
```
from w1thermsensor import W1ThermSensor, Sensor
sensor = W1ThermSensor(Sensor.DS18B20, "3c01b55615b0")
sensor.set_resolution(9)
temperature_in_celsius = sensor.get_temperature()

print(temperature_in_celsius)

```
What am I missing?
Thank you!

bsimmo, 2021-01-30:
Where di you get that code from 

 from w1thermsensor import W1ThermSensor, Sensor

there is noe Sensor in w1thermsensor.

Maybe you mean to use,
from w1thermsensor import W1ThermSensor as Sensor
to rename is as Sensor ?

timofurrer, 2021-01-30:
How did you evaluate that the w1thermsensor version you are using is in fact `2.0.0` ?

I'm almost 100% certain that the apt sources from Raspbian buster don't yet include w1thermsensor 2.0.0 and therefore `apt-get install` would install `1.3.0`.

Can you please verify that you are using w1thermsensor 2.0.0 by running this code (the same way you run your script):

```python
import w1thermsensor
print(w1thermsensor.__version__)
```

Also, see:
[image]

bsimmo, 2021-01-30:
@timofurrer my bad, I'm not up with the just released v2.
Must have a go and also use the asyncio too (perhaps put something in the docs why and when it should be used it, other than to replace my fudged threading for reading the sensors.)
thanks :-)

a-sp, 2021-01-30:
Thanks a lot for your answers @timofurrer and @bsimmo!

> How did you evaluate that the w1thermsensor version you are using is in fact `2.0.0` ?

I run the command: `(w1thermsensor --version)` and i get  `2.0.0`

Anyway, if I use your code I get that I'm on version `1.0.5`

Something fuzzy must have happened… maybe I messed with the pip installation when trying to debug the problem myself 😅

Just to be sure, what's the correct method tu upgrade?

bsimmo, 2021-01-30:
sudo apt remove python3-w1thermsensor

sudo pip3 install w1thermsensor -U

timofurrer, 2021-01-30:
In addition you may also want to use virtualenvs instead of messing with your system Python - otherwise there problems become inevitable over time ^^

closed 2021-01-30 by timofurrer

a-sp, 2021-01-30:
It worked. Thank you for taking the time to answer silly question and for your advice! 😊