timofurrer/w1thermsensor #105: Raspberry Pi zero - "sensor not ready"

mrantonSG, 2021-01-31:
**Important notices**
Before you add a new report, we ask you kindly to acknowledge the following:

[x] I have read the contributing guide lines at https://github.com/timofurrer/w1thermsensor/blob/master/.github/CONTRIBUTING.md

[x] I have read and respect the code of conduct at https://github.com/timofurrer/w1thermsensor/blob/master/.github/CODE_OF_CONDUCT.md

[x] I have searched the existing issues and I'm convinced that mine is new.

**Ask your Question**
Hi!
While w1thermsensor works flawlessly on my various Pi4 and Pi3, I struggle with it on the Pi zero (I've tried a few).
It will give me a "sensor not ready" error about 3 out of 10 times.

> File "/home/pi/.local/lib/python3.7/site-packages/w1thermsensor/core.py", line 267, in get_temperature
    raw_temperature_line = self.get_raw_sensor_strings()[1]
  File "/home/pi/.local/lib/python3.7/site-packages/w1thermsensor/core.py", line 250, in get_raw_sensor_strings
    raise SensorNotReadyError(self)
w1thermsensor.errors.SensorNotReadyError: Sensor 0317228c51ff is not yet ready to read temperature<

I can handle that as an exception in python, but I wonder what I should do differently on the zero.
I'm using the exact same setup in all cases (same SD card, same GPIO and sensor)

some more info:
Raspberry OS buster
python 3.7.3
w1thermsensor version 2.0.0 

thanks a lot!

timofurrer, 2021-02-03:
I've never tried to use it on a Pi zero and wouldn't really know what the issue could be. It might be some defective contact or instability in the w1 kernel modules. I'm pretty confident that from the Python side and this module there is nothing you can do about this.

However, I'll keep this issue open for a few more days just in case someone else had similar experiences and might be able to help you. :tada:

bsimmo, 2021-02-03:
I can tell you it's not a PiZero problem with the module, since I've used it on them since way back when.

You could try a different pin and see if that helps.
Or check contacts and power.
(this is on the Pi forum too iirc?)

Saying that, not sure if any of mine have used a recent buster image.

mrantonSG, 2021-02-03:
> I've never tried to use it on a Pi zero and wouldn't really know what the issue could be. It might be some defective contact or instability in the w1 kernel modules. I'm pretty confident that from [...]

I agree that it should have to do somehow with the pi zero since it is very reproducible. Tested it with 3 different zeros - all behaved the same. No issues on the larger Pi.

My workaround is that I read the sensor permanently all 2 seconds in its own thread and write the measurements in a variable that I read out when ever I need it. Its not time critical, so this solution is ok.

mrantonSG, 2021-02-03:
> I can tell you it's not a PiZero problem with the module, since I've used it on them since way back when.
> 
> You could try a different pin and see if that helps.
> Or check contacts and power.
> (this is on the Pi forum too iirc?)
> 
> Saying that, not sure if any of mine have used a recent buster image.

Yes - its in the Pi forum. I've put it there first, since I thought its more Pi related. I've received a comment that I should message the developer and thats why I've put it here.

The sensor sits on a custom made pcb, so switching the pin in that setup is not possible, but I'll investigate further with connecting the sensor directly and use different pins.

pbill2003, 2021-03-04:
Similar issue here fwiw. 
Same setup - pi-zero, kernel 5.4.83, w1thermsensor version 2.0.0, python 3.7.3.

File "/usr/local/lib/python3.7/dist-packages/w1thermsensor/core.py", line 267, in get_temperature
raw_temperature_line = self.get_raw_sensor_strings()[1]
File "/usr/local/lib/python3.7/dist-packages/w1thermsensor/core.py", line 250, in get_raw_sensor_strings
raise SensorNotReadyError(self)
w1thermsensor.errors.SensorNotReadyError: Sensor 3c01d6075020 is not yet ready to read temperature

These errors turn up intermittently. No pattern. I'm reading a single sensor every 10s. Sometimes it might fail after an hour or so. Sometimes after 8hrs.

FFloresM, 2021-05-01:
I got the same error
```
Traceback (most recent call last):
  File "thermsensor_lcd.py", line 71, in <module>
    temp = sensor.get_temperature() #temp en celcius
  File "/usr/local/lib/python3.7/dist-packages/w1thermsensor/core.py", line 267, in get_temperature
    raw_temperature_line = self.get_raw_sensor_strings()[1]
  File "/usr/local/lib/python3.7/dist-packages/w1thermsensor/core.py", line 250, in get_raw_sensor_strings
    raise SensorNotReadyError(self)
w1thermsensor.errors.SensorNotReadyError: Sensor 0001d607c028 is not yet ready to read temperature
```
It is a hardware issue. I think it is a wire issue. I am using the cable jumper dupont and I soldered it with tin.
The problem was in the Data cable.

Blindfreddy, 2021-11-16:
I have now tested this on Pi models 2, 3, and 4. Result:   no read errors whatsoever on any of them.
Steps to reproduce:
1. Read w1 devices on 1st generation pi - this results in many read errors as stated above
2. Move the sd card to a 2nd, 3rd or 4th generation pi -  no read errors

I agreed that there is nothing wrong with the python code.   Also, because reading the w1 devices from a bash shell rarely fails, by implication there must be some defect in the underlying python os.read code. My guess is that it doesn't wait long enough for the driver to return the result, ie. some kind of timing problem.

I raise a defect with python.

Mr-Jay-420, 2023-02-22:
I had the same problem with my Pi Zero. However, updating the firmware with rpi-update to 6.1.12 (as of today) solved this issue for me.