timofurrer/w1thermsensor #114: Weird behavior with DS18B20

sdiama, 2022-06-04:
### Erroneous Behavior

I run a "while/true loop" (every 5 seconds) to read the temperatures from multiple DS18B20 sensors.
When I disconnect the pin of one sensor from the breadboard, in order to emulate a cable/sensor problem, the "get_temperature()" method returns big negative numbers (ex. -109.9375, -110.1875) for multiple readings (10+ readings) before a "w1thermsensor.errors.SensorNotReadyError" exception is thrown.

### Expected Behavior

"w1thermsensor.errors.SensorNotReadyError" exception to be thrown as soon as the sensor disconnected

### Steps to reproduce

Unpin a DS18B20 sensor while in a while/true loop

### Version

2.0

### Python Version

3.9.2

### Operating System

Raspberry Pi

### Guidelines

- [X] I have read the contributing guide lines at [CONTRIBUTING](https://github.com/timofurrer/w1thermsensor/blob/master/.github/CONTRIBUTING.md)
- [X] I have read and respect the code of conduct at [Code of Conduct](https://github.com/timofurrer/w1thermsensor/blob/master/.github/CODE_OF_CONDUCT.md)
- [X] I have searched the existing issues and I'm convinced that mine is new

timofurrer, 2022-06-07:
Hey @sdiama ! w1thermsensor is based on the w1 therm kernel modules in a sense that it reads from the files it exposes in `/sys`. Therefore, can you check the sensor file in `/sys/bus/w1/devices` and post its output here?

I'm curious to what it will contain in your case, before it [isn't ready](https://github.com/timofurrer/w1thermsensor/blob/1c683557ca05660c95ae255544e59cd7f2328949/src/w1thermsensor/core.py#L245) any longer.

sdiama, 2022-06-12:
3 days now I'm trying to reproduce the error by removing pins, of multiple sensors in my test breadboard and I cannot get the same error! Maybe it was a sensor fault?
I made some changes to my code. If a temperature below of -20  is returned by "get_temperature()" method (they are ambient sensors), then I also read the /sys/bus/w1/devices/xxxxxxx/w1_slave and write it to a log file, in order to give you better feedback in the future.