timofurrer/w1thermsensor #115: Support bulk read feature implemented in linux driver

CJKohler, 2023-01-04:
It looks like the W1_Therm driver now supports bulk-read, which sends one command to all devices on the bus to read and convert the temperature. These can then be quickly read-out, without having to wait 750 msec for each sensor.

It looks like Bulk Read support was added in May 2020 in the linux driver:
https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/w1/slaves/w1_therm.c

`trigger_bulk_read() - function to trigger a bulk read on the bus`
Any chance you could add this to the w1thermsensor package?

timofurrer, 2023-01-08:
@CJKohler looks indeed like a nice addition. Unfortunately, I don't have enough spare time right now to work on this, but would be more than happy to review a PR and get this feature released!

then3rd, 2023-08-09:
I'd love to see this implemented. I would love to help, however it's a bit over my skillet at this time.

h-wissing, 2024-10-13:
Hello, 
I made that post in RPi-forum because I did not get a reply here from Mr. Furrer. I also addressed another issue. When using DS 2482-800 8-channel I2C to W1 converter only the first channel has the bulk read file / ability. Another problem is, that for initiation - writing "trigger /n" to the file you need supervisor rights or have to change them in bevorehand. All ohter commands can be used by standard user.

Regards

timofurrer, 2024-10-13:
I still don't have the bandwidth to implement this. However, we are open for contributions.