timofurrer/w1thermsensor #118: [Feature]: Temperature Correction/Calibration

brettlounsbury, 2023-09-09:
### Description

It seems to me that most of the DS18B20 sensors I use need to be calibrated (I use [this method](https://www.instructables.com/Calibration-of-DS18B20-Sensor-With-Arduino-UNO/)).  It would be awesome if the library could return a corrected value based on calibration data provided to it.

Would you be willing to accept a pull request where the library takes a calibration_data dict and if that data is provided, it returns corrected temperatures based on the calibration data instead of the raw values?  Alternately, I could write a second method that returns the corrected temperatures based on that calibration data if you'd prefer that.

### Guidelines

- [X] I have read the contributing guide lines at [CONTRIBUTING](https://github.com/timofurrer/w1thermsensor/blob/master/.github/CONTRIBUTING.md)
- [X] I have read and respect the code of conduct at [Code of Conduct](https://github.com/timofurrer/w1thermsensor/blob/master/.github/CODE_OF_CONDUCT.md)
- [X] I have searched the existing issues and I'm convinced that mine is new

timofurrer, 2023-09-10:
Yes, I think a new method makes sense and I'm totally open to a PR :)

brettlounsbury, 2023-09-11:
@timofurrer  - I submitted a PR here: https://github.com/timofurrer/w1thermsensor/pull/119

brettlounsbury, 2023-09-17:
Code was merged

brettlounsbury, 2023-09-17:
@timofurrer  - do you have a timeline for releasing the newest version, at which point we can close this out?

brettlounsbury, 2023-09-26:
Gentle nudge @timofurrer  - can you cut a new release with the latest changes so that they can be consumed from pypi?

timofurrer, 2023-09-27:
@brettlounsbury sorry that it took so long, release v2.3.0 just landed on PyPI: https://pypi.org/project/w1thermsensor/2.3.0

closed 2023-09-27 by timofurrer